hberntsen/mhi-ac-ctrl-esp32 #2: Question: wifi and BLE together

glsf91, 2023-02-18:
I have the software up and running with a D1 mini esp32 (without using the CS lines). So it can run on the original mhs-ac-ctrl PCB.

You wrote you have used this software with 8 BLE temp sensors.
Did also have Wifi and MQTT active in this config?
Because when I add esp32_ble_tracker to my config, the esp boots and resets right away. Without wifi and MQTT BLE is working.

hberntsen, 2023-02-18:
I am using those settings without issues:

```
esp32:
  board: esp32-c3-devkitm-1
  framework: 
    type: esp-idf
    sdkconfig_options: 
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y

esp32_ble_tracker:
  scan_parameters:
    active: false

sensor:
  - platform: pvvx_mithermometer
    mac_address: "aa:bb:cc:dd:ee:ff"
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
    battery_level:
      name: "Battery-Level"
    battery_voltage:
      name: "Battery-Voltage"
    signal_strength:
      name: "Signal"
```

Though I'm not using MQTT, just the API connection to Home Assistant.

I see you closed the issue, I just wanted to share my configuration
anyway.

glsf91, 2023-02-18:
Thanks.
Can you tell me how I activate the debug logging for MHI-AC-Ctrl-core.cpp ?
There are some ESP_LOGD messages but I don't see them. I have added in the yaml:
```
logger:
  level: DEBUG

```
But looks like this is not enough.

hberntsen, 2023-02-18:
I also have the same issue. I just renamed the relevant entries to LOGI
when I needed them. If you find the solution, please let me know :).

glsf91, 2023-02-18:
I found the solution.
Put next line ABOVE "#include "esp_log.h""
```
#define LOG_LOCAL_LEVEL ESP_LOG_DEBUG
#include "esp_log.h"
```
And in 
```
void mhi_ac_ctrl_core_init() {
    esp_log_level_set(TAG,ESP_LOG_DEBUG);
    ESP_LOGD(TAG, "This debug message");

```

Now debug is working.

hberntsen, 2024-11-24:
I think this issue has been resolved now :)

closed 2024-11-24 by hberntsen