hberntsen/mhi-ac-ctrl-esp32 #3: Sync with latest source

HpNoTiQ56, 2024-05-03:
Hi,
I just bought new esp32 mini to get bluetooth proxy and I'm discovering a lot of loop error which I didn't have with d1 mini pro. Did you update your version to enable new enhance framesize and use 3D auto and other stuff? If not, do you plan it?

Thank you,

hberntsen, 2024-05-03:
Hey! It was my planning to do so at some point, I've already refactored the code for newer esphome versions a while ago, I can push that next week. For the rest, I'd also like to have the newer features myself too. I expect my units to be new enough to support the bigger framesize. It is just one of my many side projects ;).

Does my current version of the code run stable on your units?

HpNoTiQ56, 2024-05-20:
Thank you for your answer : I'm using a wemos d1 mini and can't flash it with your code. First, i tried with your code but was not flashing (normal since another board layout is waited), then i tried to modify board by wemos_d1_mini and esp32dev with no success at compiling with latest esphome.

MrC0nst, 2024-05-27:
> Thank you for your answer : I'm using a wemos d1 mini and can't flash it with your code. First, i tried with your code but was not flashing (normal since another board layout is waited), then i tried to modify board by wemos_d1_mini and esp32dev with no success at compiling with latest esphome.

I have compiled this code for esp32 dev board with esphome 2024.5.3

https://github.com/hberntsen/mhi-ac-ctrl-esp32-c3/blob/e68566512c19821c9d168f607f340032cf7778da/esphome/my_components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp#L96
```
#define max(a,b)            (((a) > (b)) ? (a) : (b))
#define min(a,b)            (((a) < (b)) ? (a) : (b))
#define CLAMP(x, lower, upper) (min(upper, max(x, lower)))
```
and comment

hberntsen, 2024-11-24:
@HpNoTiQ56 I see I did not respond, sorry. Did you still need help?

harm-nedap, 2025-01-12:
Lager frame size, 3D auto and vanes LR is now supported!

hberntsen, 2025-01-12:
Larger frame size, 3D auto and vanes LR is now supported!

I'm closing the issue as I did not receive a response, feel free to open again though.

closed 2025-01-12 by hberntsen