hberntsen/mhi-ac-ctrl-esp32 #4: Need to fix some errors

MrC0nst, 2024-05-27:
Dear Harm Berntsen,

Could you please fix some errors?
https://github.com/hberntsen/mhi-ac-ctrl-esp32-c3/blob/e68566512c19821c9d168f607f340032cf7778da/esphome/my_components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp#L62 **GPIO_NUM_9** need to be replaced with **GPIO_CS_OUT**
same for https://github.com/hberntsen/mhi-ac-ctrl-esp32-c3/blob/e68566512c19821c9d168f607f340032cf7778da/esphome/my_components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp#L72

https://github.com/hberntsen/mhi-ac-ctrl-esp32-c3/blob/e68566512c19821c9d168f607f340032cf7778da/esphome/my_components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp#L399
What **GPIO_NUM_3** is used for?

Thanks in advance.

hberntsen, 2024-11-24:
GPIO 3 is the Red LED on the board, see https://docs.ai-thinker.com/_media/esp32/docs/esp-c3-32s-kit-v1.0_specification.pdf . It was nice during development but not really needed now. I'll get rid of it

hberntsen, 2024-11-24:
I've fixed these issues :)!

closed 2024-11-24 by hberntsen