hberntsen/mhi-ac-ctrl-esp32 #5: Vanes, pins and bugfixes

unregist, 2024-11-16:
I have made some changes to get it working for me, how I want it :)

- Fixed the vanes (more of a workaround, HA doesn't support vane settings as such)
- Fix #4 
- Customised the pins for my hardware (I can obviously exclude this commit)
- Added the MIN and MAX definitions
- Fixed some bugs 
- Added extended cv.COMPONENT_SCHEMA so I can set switch/sensors to disabled_by_default 
- renamed folder my_components to components so it can be used as an external component rather than the deprecated my_components.
- on my todo list is update the example yaml. 

Do you want me to create a pull request for these changes? If not all, let me know which ones you want to be included/excluded.

hberntsen, 2024-11-24:
Good to hear it is working for you! I saw your repo and cherry-picked some changes already, thanks!

* Vanes: I've worked on that a bit and it I got it working now. I've renamed the values to the ones used in https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/blob/de1051a67ac4a1b70d828d41ecb63afd17cca4fc/example.yaml#L272 . Does this also work for you?
* #4 is fixed now
* Do you think it would be feasible to move the pin definitions to the esphome yaml? That way we can keep it generic.
* I don't need the MIN and MAX definitions for my ESP32-C3. Are you using a ESP32-C3 as well? I'm compiling with `docker run --rm -v "${PWD}":/config -v "${PWD}/.penv:/root/.platformio/penv" -it esphome/esphome:2024.11 compile example.yaml` . That does not complain about those definitions.
* Bugfixes are always welcome. Can you create PRs for those?
* I've renamed the components folder as well now, thanks!
* example.yaml, PRs welcome!

unregist, 2024-11-24:
- This is what I have for vanes in my yaml.  I added a function called set_vanes. I couldn't figure out how it was supposed to work in the original code.

```
select:
  - platform: template
    options:
      - "Swing"
      - "Up"
      - "Up/Center"
      - "Center/Down"
      - "Down"
    name: ${device_name} Vanes
    id: fan_control_ud
    optimistic: true
    restore_value: true ## Workaround until I fix the C++ Code reporting the value properly
    on_value:
       - lambda: |-
          if (id(fan_control_ud).state == "Up") {
            return ((MhiAcCtrl*)id(${device_id}))->set_vanes("1");
          } else if (id(fan_control_ud).state == "Up/Center") {
            return ((MhiAcCtrl*)id(${device_id}))->set_vanes("2");
          } else if (id(fan_control_ud).state == "Center/Down") {
            return ((MhiAcCtrl*)id(${device_id}))->set_vanes("3");
          } else if (id(fan_control_ud).state == "Down") {
            return ((MhiAcCtrl*)id(${device_id}))->set_vanes("4");
          } else if (id(fan_control_ud).state == "Swing") {
            return ((MhiAcCtrl*)id(${device_id}))->set_vanes("Swing");
          } else {
            return;
          }

``` 

- I'd love pin definitions in yaml. I only got as far as doing the yaml config check on pins, didn't got round to having the c++ code use those though. 
- Min Max might be caused by my testing on ESP32 (non C3). I ran short on C3 modules. Put a bulk order in for black friday. Hopefully I can get rid of my remaining non-C3 modules now :)

unregist, 2024-12-04:
How are you doing the vanes? 
I can't figure out how your code is to be configured in yaml with a select, the example yaml select doesn't work.

A comment in mhi_ac_ctrl.h refers to select.py, but no such file exists. (line 114)
_Select options must match the ones in the select.py Python code_

hberntsen, 2024-12-04:
Sorry, my bad. I forgot to commit that one. Can you try again?

hberntsen, 2024-12-04:
I've added GitHub Actions to prevent this next time :)

unregist, 2024-12-04:
Sweet, that all works for me now. 
The only issue I have at this point is a temperature jitter.

I did have a hack where it only updates the temperature when the difference exceeds 0.25 degrees. 
This does delay the reporting in a temperature change but avoids the temperature updating multiple times per second (see log below). It's a bit of an eyesore on the dashboard in home assistant when it does this as it also moves the temperature gauge.

I think this might only be an issue on some AC models. I only have this issue on 2 of the 3 units.

Any idea for a more elegant fix?

<details>

<summary>Log</summary>

[09:33:24][D][climate:419]:   Current Temperature: 28.00°C
[09:33:24][D][climate:419]:   Current Temperature: 27.75°C
[09:33:24][D][climate:419]:   Current Temperature: 28.00°C
[09:33:24][D][climate:419]:   Current Temperature: 27.75°C
[09:33:25][D][climate:419]:   Current Temperature: 28.00°C
[09:33:25][D][climate:419]:   Current Temperature: 27.75°C
[09:33:26][D][climate:419]:   Current Temperature: 28.00°C
[09:33:26][D][climate:419]:   Current Temperature: 27.75°C
[09:33:28][D][climate:419]:   Current Temperature: 28.00°C
[09:33:28][D][climate:419]:   Current Temperature: 27.75°C
[09:33:28][D][climate:419]:   Current Temperature: 28.00°C
[09:33:29][D][climate:419]:   Current Temperature: 27.75°C
[09:33:32][D][climate:419]:   Current Temperature: 28.00°C
[09:33:32][D][climate:419]:   Current Temperature: 27.75°C
[09:33:32][D][climate:419]:   Current Temperature: 28.00°C
[09:33:32][D][climate:419]:   Current Temperature: 27.75°C
[09:33:33][D][climate:419]:   Current Temperature: 28.00°C
[09:33:33][D][climate:419]:   Current Temperature: 27.75°C
[09:33:33][D][climate:419]:   Current Temperature: 28.00°C
[09:33:33][D][climate:419]:   Current Temperature: 27.75°C
[09:33:33][D][climate:419]:   Current Temperature: 28.00°C
[09:33:33][D][climate:419]:   Current Temperature: 27.75°C
[09:33:33][D][climate:419]:   Current Temperature: 28.00°C
[09:33:33][D][climate:419]:   Current Temperature: 27.75°C
[09:33:35][D][climate:419]:   Current Temperature: 28.00°C
[09:33:35][D][climate:419]:   Current Temperature: 27.75°C
[09:33:35][D][climate:419]:   Current Temperature: 28.00°C
[09:33:35][D][climate:419]:   Current Temperature: 27.75°C
[09:33:35][D][climate:419]:   Current Temperature: 28.00°C
[09:33:35][D][climate:419]:   Current Temperature: 27.75°C
[09:33:35][D][climate:419]:   Current Temperature: 28.00°C
[09:33:35][D][climate:419]:   Current Temperature: 27.75°C
[09:33:36][D][climate:419]:   Current Temperature: 28.00°C
[09:33:36][D][climate:419]:   Current Temperature: 27.75°C
[09:33:37][D][climate:419]:   Current Temperature: 28.00°C
[09:33:37][D][climate:419]:   Current Temperature: 27.75°C
[09:33:37][D][climate:419]:   Current Temperature: 28.00°C
[09:33:37][D][climate:419]:   Current Temperature: 27.75°C
[09:33:37][D][climate:419]:   Current Temperature: 28.00°C
[09:33:37][D][climate:419]:   Current Temperature: 27.75°C
[09:33:37][D][climate:419]:   Current Temperature: 28.00°C
[09:33:37][D][climate:419]:   Current Temperature: 27.75°C
[09:33:37][D][climate:419]:   Current Temperature: 28.00°C
[09:33:38][D][climate:419]:   Current Temperature: 27.75°C
[09:33:38][D][climate:419]:   Current Temperature: 28.00°C
[09:33:38][D][climate:419]:   Current Temperature: 27.75°C
[09:33:39][D][climate:419]:   Current Temperature: 28.00°C
[09:33:39][D][climate:419]:   Current Temperature: 27.75°C
[09:33:39][D][climate:419]:   Current Temperature: 28.00°C
[09:33:39][D][climate:419]:   Current Temperature: 27.75°C
[09:33:39][D][climate:419]:   Current Temperature: 28.00°C
[09:33:39][D][climate:419]:   Current Temperature: 27.75°C
[09:33:39][D][climate:419]:   Current Temperature: 28.00°C
[09:33:39][D][climate:419]:   Current Temperature: 27.75°C
[09:33:40][D][climate:419]:   Current Temperature: 28.00°C
[09:33:40][D][climate:419]:   Current Temperature: 27.75°C
[... 171 lines cut ...]
[09:34:03][D][climate:419]:   Current Temperature: 27.75°C
[09:34:03][D][climate:419]:   Current Temperature: 28.00°C
[09:34:03][D][climate:419]:   Current Temperature: 27.75°C
[09:34:04][D][climate:419]:   Current Temperature: 28.00°C
[09:34:04][D][climate:419]:   Current Temperature: 27.75°C
[09:34:04][D][climate:419]:   Current Temperature: 28.00°C
[09:34:04][D][climate:419]:   Current Temperature: 27.75°C
[09:34:04][D][climate:419]:   Current Temperature: 28.00°C
[09:34:05][D][climate:419]:   Current Temperature: 27.75°C
[09:34:05][D][climate:419]:   Current Temperature: 28.00°C
[09:34:05][D][climate:419]:   Current Temperature: 27.75°C
[09:34:05][D][climate:419]:   Current Temperature: 28.00°C
[09:34:05][D][climate:419]:   Current Temperature: 27.75°C
[09:34:05][D][climate:419]:   Current Temperature: 28.00°C
[09:34:06][D][climate:419]:   Current Temperature: 27.75°C
[09:34:06][D][climate:419]:   Current Temperature: 28.00°C
[09:34:06][D][climate:419]:   Current Temperature: 27.75°C
[09:34:06][D][climate:419]:   Current Temperature: 28.00°C
[09:34:07][D][climate:419]:   Current Temperature: 27.75°C
[09:34:07][D][climate:419]:   Current Temperature: 28.00°C
[09:34:07][D][climate:419]:   Current Temperature: 27.75°C
[09:34:07][D][climate:419]:   Current Temperature: 28.00°C
[09:34:08][D][climate:419]:   Current Temperature: 27.75°C
[09:34:08][D][climate:419]:   Current Temperature: 28.00°C
</details>

hberntsen, 2024-12-06:
I've seen oscillations too, though not so often. 

ESPHome has nice filters, see https://esphome.io/components/sensor/#sensor-filters . But they only seem to apply to sensors. I've asked on their discord for ideas: https://discord.com/channels/429907082951524364/524177279270780928/1314589160010416139

hberntsen, 2024-12-23:
@unregist The discord suggested to add a sensor to apply the filters on. I've implemented that in https://github.com/hberntsen/mhi-ac-ctrl-esp32-c3/commit/eb1489a3204f845e7b7dff59d4d55dac81596961

unregist, 2024-12-23:
What a nice birthday present to wake up to. Thanks!

A few things I changed in my config.

- I changed the heartbeat filter to throttle so you don't have an entry in the log every so many seconds, otherwise you get the same result.

- I removed the round filter in my setup, so I can show decimals in Home Assistant Climate, just add temperature_step into your ESPHome yaml:

```
climate:
  - platform: MhiAcCtrl
    mhi_ac_ctrl_id: ${mhi_device_id}
    id: ${device_id}
    name: "${device_name}"
    visual:
      temperature_step:
        target_temperature: 1.0
        current_temperature: 0.1
```

hberntsen, 2025-01-12:
Haha, happy belated birthday!

Throttle is described as follows:
> Throttle the incoming values. When this filter gets an incoming value, it checks if the last incoming value is at least `specified time period` old. If it is not older than the configured value, the value is not passed forward.
-- https://esphome.io/components/sensor/#throttle

Thus, when the oscillations stop, you might not see the value it settled on due to the throttle. The heartbeat does regular updates, preventing this problem. The climate component only sends a new temperature when it has actually changed. I'm running logs at INFO so I don't see the updates being logged.

It would be even nicer to use something like throttle_average. The problem I see with that filter is that it only averages each incoming temperature value, not taking into account how long that value was present.

I didn't know about the temperature_step, thanks!

closed 2025-01-16 by unregist