hberntsen/mhi-ac-ctrl-esp32 #10: Allow feeding in a 'light ID' instead of a pin number for the status LED

oliv3r, 2025-01-15:
It would be much nicer if we would use a id(light) instead of pin for the status led. This would allow us to use the onboard `status_led` (if configured) e.g.

```yaml
light:
  - platform: status_led
    name: "Status LED"
    id: board_status_led
    pin:
      number: GPIO02
      ignore_strapping_warning: true
    entity_category: diagnostic
```

But also a regular LED (e.g. not the status_led), if an additional LED is available.

This would allow for keeping the configuration in the high-level ESPHome yaml file instead.

oliv3r, 2025-01-15:
You misunderstand :)

The status led is indeed a common esphome function, however you currently have
```yaml
status_led:
  pin:
    #white led, off when all good
    number: 19
```

Instead, if we could tell MHI-AC-Ctrl to use 'id(status_led)` we could use a single led for multiple functions (e.g. when WiFi is not connected, it's blinking, if the API is not connected, it's blinking. If MHI-AC-Ctrl says things are not okay ... it could be blinking.

```
The status_led light platform allows to share a single LED for indicating the status of the device (when on error/warning state) or as binary light (when on OK state). This is useful for devices with only one LED available. You can also use a binary [Output Component](https://esphome.io/components/output/#output).
```

But being able to use any `id(light)` as status_led input to MHI-ctrl, we could configure a regular GPIO output as LED, and tell MHI-AC-Ctrl to use that intead. I'm not sure if you can have several `status_led`'s or if they all blink in sync as you can only have one `status_led` and the others would need to be 'regular lights'

hberntsen, 2025-01-15:
Ah, I see. In my units, the module is hidden inside the AC unit. If I turn on the LED, I can barely see it. After running for over more than a year, the only problem I encountered is related to the WiFi signal. Frame errors are very rare, it just works.

You could use the frame errors sensor to trigger an LED, i.e. do a

```
    frame_errors:
      name: ${device_name} Frame errors
      on_value_range:
        - above: 10
          then:
            - switch.turn_on: error led
```

Would that cover your use case?

oliv3r, 2025-01-15:
I was just looking into removing 'component specifics' and moving them into 'generic ESPHome-yaml' methods.

From what I've found so far (my search isn't finished), you can use https://esphome.io/api/component_8cpp_source#l00151 to print a warning, (likewise to https://esphome.io/api/component_8cpp_source#l00159)

HOWEVER, I may have just realized the error of my ways. I'm so used to using `light: platform: status_led` that I didn't realize you ARE using the `status_led`. The comment 'white led` through me off I suppose, as I tend to use the ESP modules led as status led (usually the blue led on pin2).

So this is not relevant anymore :)

closed 2025-01-15 by oliv3r

hberntsen, 2025-01-15:
That's okay! I didn't know about the `status_has` functions. Nice! I don't have a use case yet myself since my modules work reliably

hberntsen, 2025-01-15:
AC communications failure should log `Snapshot timeout`: https://github.com/hberntsen/mhi-ac-ctrl-esp32-c3/blob/3d18a753f22891464e2fea17ee0aceae479141f8/esphome/components/MhiAcCtrl/mhi_ac_ctrl.h#L267C28-L267C44 . Hope you get it to work :)!

hberntsen, 2025-01-15:
I've replaced that warning log with the  `status_set_warning` when no frame is received. So it should signal warning status and log a message that there is no communications. That was easy to replace.