hberntsen/mhi-ac-ctrl-esp32 #11: Allow feeding the SPI slave in from the ESPHome yaml

oliv3r, 2025-01-15:
Being able to feed in an ESPHome yaml SPI slave does require the slave to be available in esphome/platform IO first. Having that however would allow for #9

hberntsen, 2025-01-15:
What is your practical use case of sharing the SPI bus between several devices? In this case we are a slave. So the AC is going to determine by itself when it is going to send data, which is quite often (iirc every 40ms for the 20 bytes frame size).

oliv3r, 2025-01-15:
None particular right now, just that it would be following the cleaner ESPHome yaml way of configuring things, e.g. declaring a (slave) bus, and then attaching a device. That way, the entire SPI bus setup, quirck handling etc, would be 'one layer higher' rather then having to 're-do' all this in the MHI-AC-Ctrl module.

hberntsen, 2025-01-15:
Yeah I agree. Once esphome provides a SPI slave thing I'm open to implementing that. Of course provided that it gives us the functionality we need

hberntsen, 2025-03-21:
Let's create a new issue once the SPI slave support is there

closed 2025-03-21 by hberntsen