hberntsen/mhi-ac-ctrl-esp32 #12: ESP32 not working? (WDT reset)

oliv3r, 2025-01-15:
I've built with `logger: level: DEBUG` but I'm not getting anything useful ...

```
[C][status_led:056]: Setting up Status LED...
[D][esp-idf:000]: I (952) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 

[C][light:021]: Setting up light 'Status LED'...
[D][status_led:039]: 'Status LED': Setting initial state
[D][status_led:050]: 'Status LED': Setting state OFF
[D][light:036]: 'Status LED' Setting:
[D][light:041]:   Color mode: 
[D][text_sensor:064]: 'ESPHome Version': Sendiets Jul 29 2019 12:21:46
[18:43:43]
[18:43:43]rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[18:43:43]configsip: 0, SPIWP:0xee
[18:43:43]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[18:43:43]mode:DIO, clock div:2
[18:43:43]load:0x3fff0030,len:7112
[18:43:43]load:0x40078000,len:15624
[18:43:43]load:0x40080400,len:4
[18:43:43]load:0x40080404,len:3876
[18:43:43]entry 0x4008064c
[18:43:43]I (58) boot: ESP-IDF 5.1.5 2nd stage bootloader
[18:43:43]I (58) boot: compile time Jan 15 2025 16:18:21
[18:43:43]I (58) boot: Multicore bootloader
[18:43:43]I (63) boot: chip revision: v3.1
[18:43:43]I (67) boot.esp32: SPI Speed      : 40MHz
[18:43:43]I (71) boot.esp32: SPI Mode       : DIO
[18:43:43]I (76) boot.esp32: SPI Flash Size : 4MB
[18:43:43]W (80) boot.esp32: PRO CPU has been reset by WDT.
[18:43:43]W (86) boot.esp32: WDT reset info: PRO CPU PC=0x4008cc6c
[18:43:43]W (92) boot.esp32: WDT reset info: APP CPU PC=0x4008cc1b
```

is MHI-AC-Ctrl expected to tickle the watchdog? Note that I have not yet connected the AC unit ...

Things only started to fail once I added the MHI in my yaml

```yaml
substitutions:
  name: "mhi-ac-ctrl"
  friendly_name: "MHI-AC-Ctrl"
  device_description: "Mitsubishi Heavy Industries Air Conditioner Controller"

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  comment: ${device_description}
  name_add_mac_suffix: false #remove
  min_version: 2024.6.6
  project: 
    name: oliver.${name}
    version: "3.0"

esp32:
  board: mhetesp32minikit
  framework:
    type: esp-idf

external_components:
  - source: github://hberntsen/mhi-ac-ctrl-esp32-c3/esphome@master
    components: [ MhiAcCtrl ]

logger:
  level: DEBUG

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    password: !secret fallback_hotspot_password

captive_portal:

improv_serial:

api:
  encryption:
    key: !secret api_encryption_key

ota:
  - platform: esphome
    password: !secret ota_password

web_server:
  port: 80
  auth:
    username: !secret web_server_user
    password: !secret web_server_password

time:
  - platform: homeassistant

i2c:
  sda: GPIO21
  scl: GPIO22

text_sensor:
  - platform: version
    name: "ESPHome Version"
    hide_timestamp: true

  - platform: wifi_info
    ip_address:
      name: IP
    mac_address:
      name: MAC
    ssid:
      name: SSID
    bssid:
      name: BSSID

light:
  - platform: status_led
    name: "Status LED"
    id: board_status_led
    pin:
      number: GPIO02
      ignore_strapping_warning: true
    entity_category: diagnostic

MhiAcCtrl:
  id: mhi_ac_device
  mosi_pin: GPIO23
  miso_pin: GPIO19
  sclk_pin: GPIO18

sensor:
  - platform: uptime
    name: "Uptime"

  - platform: wifi_signal
    name: "Signal Strength"
    update_interval: 60s

  - platform: bme280_i2c
    address: 0x76
    temperature:
      name: "Temperature"
    pressure:
      name: "Pressure"
    humidity:
      name: "Humidity"

  - platform: MhiAcCtrl
    mhi_ac_ctrl_id: mhi_ac_device
    total_energy:
      name: "Total Energy"
      filters:
        - throttle: 60s
    power:
      name: "Power"
    frame_errors:
      name: "Frame errors"
    climate_current_temperature:
      id: climate_current_temperature
      filters:
        - heartbeat: 10s
        - round: 0

select:
  - platform: MhiAcCtrl
    mhi_ac_ctrl_id: mhi_ac_device
    vanes_ud:
      name: "Vanes Up/Down"
    vanes_lr:
      name: "Vanes Left/Right"

switch:
  - platform: MhiAcCtrl
    mhi_ac_ctrl_id: mhi_ac_device
    active_mode:
      name: "Active mode"
```

hberntsen, 2025-01-15:
The code is not doing anything with the watchdog, nor should it completely block the CPU. The MHI loop runs in its own task, which should block just that task. The ESPHome climate component does do a blocking wait on a new frame, but that's limited to 100ms and ESPHome will simply report that it is taking too long but carry on just fine.

I'm not seeing any pin configuration for the SPI CS loopback, which by default is from pin 9 to 10. Is that the same on your board?

hberntsen, 2025-01-15:
Oh, and what's up with the esphome version? I see `[D][text_sensor:064]: 'ESPHome Version': Sendiets Jul 29 2019 12:21:46` in your logs? 2019 sounds old, I'm running 2024.12.2.

unregist, 2025-01-16:
The defaults for cs_in and cs_out are GPIO9 and GPIO10 which is fine for the ESP32-C3, however with the ESP32 it may conflict as they could be used for the SPI flash.

Try overriding cs_in and cs_out in your config to something else, the pins only need to be connected to each other, not to anything else.

On my ESP32 I've used 26 and 27 for this as they are next to each other on my devboard so I just joined them with a blob of solder.

oliv3r, 2025-01-16:
> The defaults for cs_in and cs_out are GPIO9 and GPIO10 which is fine for the ESP32-C3, however with the ESP32 it may conflict as they could be used for the SPI flash.
>
> Try overriding cs_in and [...]

Hmm, curious of this sharing of pins is also true on the minikit, but if that is the case, it could be that toggling those pins would then mess with flash, causing the wdt timeout.

closed 2025-03-21 by hberntsen