hberntsen/mhi-ac-ctrl-esp32 #13: Older AC unit which does not support 33 frame length not working

unregist, 2025-01-16:
I've got 3 AC units, 2 new models, which work fine, 1 older model which doesn't support 33 frame lenths.
The older unit no longer works since recent changes.

The first commit where it is broken is 3e90be9c48d3a5072bb8afda02433819bb4c5223
It seems to be able to read data but any commands sent are not actioned by the AC unit.
Additionally it goes haywire on the Left Right vanes in the log. 

Is there a way to force it to a 20 frame length? 

<details>
  <summary>Full Log file</summary>
  
```
[11:17:37][I][app:100]: ESPHome version 2024.12.2 compiled on Jan 16 2025, 11:17:06
[11:17:37][C][wifi:600]: WiFi:
[11:17:37][C][wifi:428]:   Local MAC: D8:3B:DA:11:78:48
[11:17:37][C][wifi:433]:   SSID: [redacted]
[11:17:37][C][wifi:436]:   IP Address: 192.168.52.82
[11:17:37][C][wifi:439]:   BSSID: [redacted]
[11:17:37][C][wifi:441]:   Hostname: 'ac-living-espidf'
[11:17:37][C][wifi:443]:   Signal strength: -40 dB ▂▄▆█
[11:17:37][C][wifi:447]:   Channel: 11
[11:17:37][C][wifi:448]:   Subnet: 255.255.255.0
[11:17:37][C][wifi:449]:   Gateway: 192.168.52.1
[11:17:37][C][wifi:450]:   DNS1: 0.0.0.0
[11:17:37][C][wifi:451]:   DNS2: 0.0.0.0
[11:17:37][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Swing (index 7)
[11:17:37][C][logger:185]: Logger:
[11:17:37][C][logger:186]:   Level: DEBUG
[11:17:37][C][logger:188]:   Log Baud Rate: 0
[11:17:37][C][logger:189]:   Hardware UART: USB_SERIAL_JTAG
[11:17:37][C][logger:193]:   Level for 'component': ERROR
[11:17:37][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Left (index 0)
[11:17:37][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Swing (index 7)
[11:17:37][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Left (index 0)
[11:17:37][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Swing (index 7)
[11:17:37][C][uptime.sensor:033]: Uptime Sensor 'ac-living-espidf Uptime'
[11:17:37][C][uptime.sensor:033]:   Device Class: 'duration'
[11:17:37][C][uptime.sensor:033]:   State Class: 'total_increasing'
[11:17:37][C][uptime.sensor:033]:   Unit of Measurement: 's'
[11:17:37][C][uptime.sensor:033]:   Accuracy Decimals: 0
[11:17:37][C][uptime.sensor:033]:   Icon: 'mdi:timer-outline'
[11:17:37][C][uptime.sensor:034]:   Type: Seconds
[11:17:37][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Left (index 0)
[11:17:37][C][bluetooth_proxy:091]: Bluetooth Proxy:
[11:17:37][C][bluetooth_proxy:092]:   Active: YES
[11:17:37][C][bluetooth_proxy:093]:   Connections: 3
[11:17:37][C][bluetooth_proxy:094]:   Raw advertisements: NO
[11:17:37][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Swing (index 7)
[11:17:37][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Left (index 0)
[11:17:37][C][restart:068]: Restart Switch 'ac-living-espidf Module Restart'
[11:17:37][C][restart:070]:   Icon: 'mdi:restart'
[11:17:37][C][restart:090]:   Restore Mode: always OFF
[11:17:37][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Swing (index 7)
[11:17:37][C][esp32_ble:418]: ESP32 BLE:
[11:17:37][C][esp32_ble:419]:   MAC address: D8:3B:DA:11:78:4A
[11:17:37][C][esp32_ble:421]:   IO Capability: none
[11:17:37][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Left (index 0)
[11:17:37][C][esp32_ble_tracker:661]: BLE Tracker:
[11:17:37][C][esp32_ble_tracker:662]:   Scan Duration: 300 s
[11:17:37][C][esp32_ble_tracker:663]:   Scan Interval: 320.0 ms
[11:17:37][C][esp32_ble_tracker:664]:   Scan Window: 30.0 ms
[11:17:37][C][esp32_ble_tracker:665]:   Scan Type: ACTIVE
[11:17:37][C][esp32_ble_tracker:666]:   Continuous Scanning: True
[11:17:37][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Swing (index 7)
[11:17:37][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Left (index 0)
[11:17:37][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Swing (index 7)
[11:17:37][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Left (index 0)
[11:17:37][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Swing (index 7)
[11:17:37][C][captive_portal:089]: Captive Portal:
[... 82 lines cut ...]
[11:17:39][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Left (index 0)
[11:17:39][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Swing (index 7)
[11:17:39][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Left (index 0)
[11:17:40][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Swing (index 7)
[11:17:40][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Left (index 0)
[11:17:40][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Swing (index 7)
[11:17:40][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Left (index 0)
[11:17:40][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Swing (index 7)
[11:17:40][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Left (index 0)
[11:17:40][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Swing (index 7)
[11:17:40][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Left (index 0)
[11:17:40][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Swing (index 7)
[11:17:40][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Left (index 0)
[11:17:40][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Swing (index 7)
[11:17:40][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Left (index 0)
[11:17:40][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Swing (index 7)
[11:17:40][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Left (index 0)
[11:17:40][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Swing (index 7)
[11:17:40][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Left (index 0)
[11:17:40][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Swing (index 7)
[11:17:40][D][select:015]: 'ac-living-espidf Vanes Left Right': Sending state Left (index 0)
```

  
</details>

harm-nedap, 2025-01-16:
The left right vanes require the longer frame length. It's best to remove the option for it in the esphome yaml.

Why doesn't the commit you link not work? That one does not actually use the longer frames yet.

Your log indicates you are running something newer. Could you try setting this to false?

https://github.com/hberntsen/mhi-ac-ctrl-esp32-c3/blob/19ef7461371c5bf54a2211e086cc6066fd236184/esphome/components/MhiAcCtrl/MHI-AC-Ctrl-core.h#L12C1-L12C41

If that works we could make it an option in the yaml.

hberntsen, 2025-01-16:
The left right vanes require the longer frame length. It's best to remove the option for it in the esphome yaml.

Why doesn't the commit you link not work? That one does not actually use the longer frames yet.

Your log indicates you are running something newer. Could you try setting this to false?

https://github.com/hberntsen/mhi-ac-ctrl-esp32-c3/blob/19ef7461371c5bf54a2211e086cc6066fd236184/esphome/components/MhiAcCtrl/MHI-AC-Ctrl-core.h#L12C1-L12C41

If that works we could make it an option in the yaml.

unregist, 2025-01-16:
I ran a bunch of different commits during testing, I might have mixed them up.

Now running on the latest code with USE_LONG_FRAME set to false and that seems to fix it! thanks :)

hberntsen, 2025-01-17:
Cool! I've made it a setting in the yaml now: https://github.com/hberntsen/mhi-ac-ctrl-esp32-c3/commit/e652862f3ce4edf1cc4b9d534ef6292febe722e7

unregist, 2025-01-17:
Starting commit 312ff23567241c6097b11e94b0dcdaf102ec44b3 i get panics. 
```
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
```

Commit 2c57eafc675f207c1b4714ebe1118dcea8e6f515 is working fine

<details>
<summary>Full Log Dump</summary>

```
[08:59:48]ESP-ROM:esp32c3-api1-20210207
[08:59:48]Build:Feb  7 2021
[08:59:48]rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
[08:59:48]Saved PC:0x40048b82
[08:59:48]SPIWP:0xee
[08:59:48]mode:DIO, clock div:1
[08:59:48]load:0x3fcd5810,len:0x1684
[08:59:48]load:0x403cc710,len:0x928
[08:59:48]load:0x403ce710,len:0x2d7c
[08:59:48]entry 0x403cc710
[08:59:48]I (35) boot: ESP-IDF 4.4.5 2nd stage bootloader
[08:59:48]I (35) boot: compile time 09:45:30
[08:59:48]I (35) boot: chip revision: v0.4
[08:59:48]I (37) boot.esp32c3: SPI Speed      : 80MHz
[08:59:48]I (42) boot.esp32c3: SPI Mode       : DIO
[08:59:48]I (46) boot.esp32c3: SPI Flash Size : 4MB
[08:59:48]I (51) boot: Enabling RNG early entropy source...
[08:59:48]I (57) boot: Partition Table:
[08:59:48]I (60) boot: ## Label            Usage          Type ST Offset   Length
[08:59:48]I (67) boot:  0 otadata          OTA data         01 00 00009000 00002000
[08:59:48]I (75) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[08:59:48]I (82) boot:  2 app0             OTA app          00 10 00010000 001c0000
[08:59:48]I (90) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[08:59:48]I (97) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[08:59:48]I (105) boot: End of partition table
[08:59:48]I (109) esp_image: segment 0: paddr=00010020 vaddr=3c130020 size=5dc40h (384064) map
[08:59:48]I (175) esp_image: segment 1: paddr=0006dc68 vaddr=3fc96400 size=023b0h (  9136) load
[08:59:48]I (177) esp_image: segment 2: paddr=00070020 vaddr=42000020 size=12e83ch (1239100) map
[08:59:48]I (367) esp_image: segment 3: paddr=0019e864 vaddr=3fc987b0 size=00fa4h (  4004) load
[08:59:48]I (368) esp_image: segment 4: paddr=0019f810 vaddr=40380000 size=16248h ( 90696) load
[08:59:48]I (395) boot: Loaded app from partition at offset 0x10000
[08:59:48]I (396) boot: Disabling RNG early entropy source...
[08:59:48]I (396) cpu_start: Unicore app
[08:59:48]I (400) cpu_start: Pro cpu up.
[08:59:48]I (415) cpu_start: Pro cpu start user code
[08:59:48]I (415) cpu_start: cpu freq: 160000000 Hz
[08:59:48]I (415) cpu_start: Application information:
[08:59:48]I (418) cpu_start: Project name:     ac-living
[08:59:48]I (424) cpu_start: App version:      2024.12.4
[08:59:48]I (429) cpu_start: Compile time:     Jan 18 2025 08:33:44
[08:59:48]I (435) cpu_start: ELF file SHA256:  d6d9608c48a72703...
[08:59:48]I (441) cpu_start: ESP-IDF:          5.1.5
[08:59:48]I (445) cpu_start: Min chip rev:     v0.3
[08:59:48]I (450) cpu_start: Max chip rev:     v1.99 
[08:59:48]I (455) cpu_start: Chip rev:         v0.4
[08:59:48]I (460) heap_init: Initializing. RAM available for dynamic allocation:
[08:59:48]I (467) heap_init: At 3FCA1960 len 0001E6A0 (121 KiB): DRAM
[08:59:48]I (473) heap_init: At 3FCC0000 len 0001C710 (113 KiB): DRAM/RETENTION
[08:59:48]I (480) heap_init: At 3FCDC710 len 00002950 (10 KiB): DRAM/RETENTION/STACK
[08:59:48]I (488) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
[08:59:48]I (495) spi_flash: detected chip: generic
[08:59:48]I (499) spi_flash: flash io: dio
[08:59:48]I (503) sleep: Configure to isolate all GPIO pins in sleep state
[08:59:48]I (510) sleep: Enable automatic switching of GPIO sleep configuration
[08:59:48]I (517) coexist: coex firmware version: 27d8387
[08:59:48]I (522) coexist: coexist rom version 9387209
[08:59:48]I (527) app_start: Starting scheduler on CPU0
[08:59:48]I (532) main_task: Started on CPU0
[08:59:48]I (536) main_task: Calling app_main()
[... 380 lines cut ...]
[08:59:54]3fca9e50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[08:59:54]3fca9e70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3f000000 0x00000018
[08:59:54]3fca9e90: 0x05500008 0x00000000 0x3fcab1ac 0x676e6976 0x6f6f5220 0x3fca9e88 0x00000010 0x00000000
[08:59:54]3fca9eb0: 0x3fca9d38 0x00000000 0x00000000 0x00000020 0x41204334 0x6f637269 0x694c206e 0x676e6976
[08:59:54]3fca9ed0: 0x6f6f5220 0x3a6d006d 0x5b1b0a20 0x3fca9eb8 0x00000048 0x3c134a18 0x00000002 0x7fc00000
[08:59:54]3fca9ef0: 0x3c134970 0x0001c200 0x3fca9f30 0x00000062 0x00000200 0x00000000 0x00000000 0x3fcaa9d4
[08:59:54]3fca9f10: 0x3fcaa9f0 0x3fcaa9f0 0x00000000 0x00000000 0x00000000 0x305b1b00 0x3fca9d38 0x00000204
[08:59:54]3fca9f30: 0x3b305b1b 0x5b6d3633 0x735b5d44 0x63656c65 0x31303a74 0x203a5d35 0x2d636127 0x6976696c
[08:59:54]3fca9f50: 0x682d676e 0x7362626f 0x6e615620 0x55207365 0x6f442070 0x3a276e77 0x6e655320 0x676e6964
[08:59:54]3fca9f70: 0x61747320 0x53206574 0x49206565 0x65522052 0x65746f6d 0x6e692820 0x20786564 0x5b1b2936
[08:59:54]3fca9f90: 0x63006d30 0x61727563 0x5b1b7963 0x6d006d30 0x305b1b0a 0x1b30006d 0x0a6d305b 0x6d305b1b
[08:59:54]3fca9fb0: 0x305b1b00 0x1b00006d 0x0a6d305b 0x6d305b1b 0xbe9d2d00 0xc1574266 0x14f42332 0x48f2e61d
[08:59:54]3fca9fd0: 0xe9de1494 0x633dfcc8 0x69d059ec 0xfe148dd6 0x8163dae7 0xc49372f6 0xa522fbe1 0x258a108d
[08:59:54]3fca9ff0: 0xf4958b81 0xfb1d1b30 0xba7fa547 0x977cdce8 0x76ffd8db 0x6c16ea62 0x99422239 0x99df4c98
[08:59:54]3fcaa010: 0x5d66e4fc 0x82e56f5c 0x15097acc 0xa72c4efa 0x4cbb25a4 0xa9adb89f 0x32d23b58 0x3517f3df
[08:59:54]3fcaa030: 0xa1e91a42 0xa2ec5ade 0x0e4ce284 0x1e49c413 0xac3ed71c 0xc3404362 0xe04fa2b4 0x741dc3f8
[08:59:54]3fcaa050: 0x66af357b 0xc1d94535 0xfce9db59 0x4471ae10 0x83612572 0x0408f391 0xfdbe6761 0x81c1cdaf
[08:59:54]
[08:59:54]
[08:59:54]
[08:59:54]ELF file SHA256: d6d9608c48a72703
[08:59:54]
[08:59:54]Rebooting...
```
</details>

hberntsen, 2025-01-18:
One or two frame errors at the start is normal on my units. It might have something to do with switching over frame sizes, though I didn't research that.

It seems like once it gets a valid frame it goes into the loop() in mhi_ac_ctrl.h, updates the total energy sensor (first one it does) and somehow fails after that. After the total energy sensor it is also the moment it starts touching the snapshot data, in which I made an oopsie during the refactoring: https://github.com/hberntsen/mhi-ac-ctrl-esp32-c3/commit/a2147ff3e769f18367ce5996b1a671adaa0db908 . Even though you report that refactoring works fine, could you test with this commit? That feels more likely to cause issues than the config refactoring.

Note that I tested the code on my units, none of them failed. But the C3 has a different CPU instruction set, so it's quite different.

unregist, 2025-01-18:
Which devices are you using? These panics are on a ESP32-C3. The esp32-c3-devkitm-1 board to be specific.

The last commit a2147ff3e769f18367ce5996b1a671adaa0db908 still panics for me.

hberntsen, 2025-01-18:
I could have seen that in your logs indeed. I'm using the same.

Well, rookie mistake it seems. After removing the power sensor I could replicate the crashes. Looks like by adding the constructor for the config we lost the zero- initialisation of the sensor pointers. Since I was using all the sensors in my configs they were all overwritten so I was not affected. Commit https://github.com/hberntsen/mhi-ac-ctrl-esp32-c3/commit/9ad5395f60a9ef53e5e3e6155af323e62cb7ba3b should fix the problem for you!

unregist, 2025-01-18:
Yes. That did it 👍

closed 2025-01-18 by unregist