hberntsen/mhi-ac-ctrl-esp32 #15: MHI SRK20ZS-S: Repeated "error 1", many frame errors and sending commands not possible

nohum, 2025-02-27:
Hi!

First of all, thank you for your efforts of porting this to ESP32! :)

I am using [this board from nodo-shop](https://www.nodo-shop.nl/en/our-products/228-espac-wifi-mitsubishi-heavy-industries-interface.html), but instead of using it with D1 mini's I have put the pin-compatible [Wemos ESP32-C3 mini](https://www.wemos.cc/en/latest/c3/c3_mini.html) on it.

For reference, after studying the board layout used by nodo-shop (this should the one from [here](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Hardware.md)) I came up with the following pin mapping to make it work:

```
MhiAcCtrl:
  id: ${mhi_device_id}
  use_long_frame: true
  mosi_pin: GPIO4
  miso_pin: GPIO0
  sclk_pin: GPIO1
  cs_in_pin: GPIO5
  cs_out_pin: GPIO6
``

I have modified the board in such a way that GPIO5 and GPIO6 are connected for chip select.

The indoor HVAC model I have at home is a MHI SRK20ZS-S, when turning on everything via the breaker, I get a few frame errors (and a repeated error 1, see below), but the target temperature is read correctly. When turning on the indoor unit, chaning the target temperature, changing the fan or HVAC mode, this is correctly detected in esphome (with some ~5 seconds delay).

However, when turning on active mode, the frame error counter is incrementing rapidly and the logs are flooded with "error 1" messages looking like this:

```
[17:16:36][D][switch:012]: 'Active mode' Turning ON.
[17:16:36][D][switch:055]: 'Active mode': Sending state ON
[17:16:36][D][sensor:093]: 'Power': Sending state 5.21569 W with 0 decimals of accuracy
[17:16:36][D][esp-idf:000][mhi_task]: E (2181410) MHI-AC-CTRL-core: Missed 20 frames

[17:16:36][D][sensor:093]: 'Frame errors': Sending state 6745.00000  with 0 decimals of accuracy
[17:16:36][D][esp-idf:000][mhi_task]: W (2181457) MHI-AC-CTRL-core: error 1 . trans len: 163

[17:16:36][D][sensor:093]: 'Frame errors': Sending state 6746.00000  with 0 decimals of accuracy
[17:16:37][D][esp-idf:000][mhi_task]: E (2182460) MHI-AC-CTRL-core: Missed 20 frames

[17:16:37][D][sensor:093]: 'Frame errors': Sending state 6766.00000  with 0 decimals of accuracy
[17:16:37][D][esp-idf:000][mhi_task]: W (2182507) MHI-AC-CTRL-core: error 1 . trans len: 167

[17:16:37][D][sensor:093]: 'Frame errors': Sending state 6767.00000  with 0 decimals of accuracy
[17:16:38][D][esp-idf:000][mhi_task]: E (2183510) MHI-AC-CTRL-core: Missed 20 frames

[17:16:38][D][sensor:093]: 'Frame errors': Sending state 6787.00000  with 0 decimals of accuracy
[17:16:38][D][esp-idf:000][mhi_task]: W (2183557) MHI-AC-CTRL-core: error 1 . trans len: 168

[17:16:38][D][sensor:093]: 'Frame errors': Sending state 6788.00000  with 0 decimals of accuracy
[17:16:39][D][esp-idf:000][mhi_task]: E (2184560) MHI-AC-CTRL-core: Missed 20 frames

[17:16:39][D][sensor:093]: 'Frame errors': Sending state 6808.00000  with 0 decimals of accuracy
[17:16:39][D][esp-idf:000][mhi_task]: W (2184607) MHI-AC-CTRL-core: error 1 . trans len: 168

[17:16:39][D][sensor:093]: 'Frame errors': Sending state 6809.00000  with 0 decimals of accuracy
[17:16:40][D][esp-idf:000][mhi_task]: E (2185609) MHI-AC-CTRL-core: Missed 20 frames

[17:16:40][D][sensor:093]: 'Frame errors': Sending state 6829.00000  with 0 decimals of accuracy
[17:16:40][D][sensor:093]: 'climate_current_temperature': Sending state 23.00000 °C with 0 decimals of accuracy
[17:16:40][D][esp-idf:000][mhi_task]: W (2185657) MHI-AC-CTRL-core: error 1 . trans len: 167

[17:16:40][D][sensor:093]: 'Frame errors': Sending state 6830.00000  with 0 decimals of accuracy
[17:16:41][D][esp-idf:000][mhi_task]: E (2186659) MHI-AC-CTRL-core: Missed 20 frames

[17:16:41][D][sensor:093]: 'Frame errors': Sending state 6850.00000  with 0 decimals of accuracy
[17:16:41][D][esp-idf:000][mhi_task]: W (2186707) MHI-AC-CTRL-core: error 1 . trans len: 168

[17:16:41][D][sensor:093]: 'Frame errors': Sending state 6851.00000  with 0 decimals of accuracy
[17:16:42][D][esp-idf:000][mhi_task]: E (2187709) MHI-AC-CTRL-core: Missed 20 frames

[17:16:42][D][sensor:093]: 'Frame errors': Sending state 6871.00000  with 0 decimals of accuracy
[17:16:42][D][esp-idf:000][mhi_task]: W (2187757) MHI-AC-CTRL-core: error 1 . trans len: 153

[17:16:42][D][sensor:093]: 'Frame errors': Sending state 6872.00000  with 0 decimals of accuracy
[17:16:43][D][esp-idf:000][mhi_task]: E (2188759) MHI-AC-CTRL-core: Missed 20 frames

[17:16:43][D][sensor:093]: 'Frame errors': Sending state 6892.00000  with 0 decimals of accuracy
[17:16:43][D][esp-idf:000][mhi_task]: W (2188807) MHI-AC-CTRL-core: error 1 . trans len: 169

[17:16:43][D][sensor:093]: 'Frame errors': Sending state 6893.00000  with 0 decimals of accuracy
[17:16:44][D][esp-idf:000][mhi_task]: E (2189809) MHI-AC-CTRL-core: Missed 20 frames

[17:16:44][D][sensor:093]: 'Frame errors': Sending state 6913.00000  with 0 decimals of accuracy
[17:16:44][D][esp-idf:000][mhi_task]: W (2189857) MHI-AC-CTRL-core: wrong MOSI signature. 0x98 0x01 0x24

[17:16:44][D][esp-idf:000][mhi_task]: W (2189859) MHI-AC-CTRL-core: error -1 . trans len: 160

[17:16:44][D][sensor:093]: 'Frame errors': Sending state 6914.00000  with 0 decimals of accuracy
[17:16:45][D][esp-idf:000][mhi_task]: E (2190862) MHI-AC-CTRL-core: Missed 20 frames

[17:16:45][D][sensor:093]: 'Frame errors': Sending state 6934.00000  with 0 decimals of accuracy
[17:16:45][D][esp-idf:000][mhi_task]: W (2190907) MHI-AC-CTRL-core: error 1 . trans len: 168

[17:16:45][D][sensor:093]: 'Frame errors': Sending state 6935.00000  with 0 decimals of accuracy
[17:16:46][D][climate:011]: 'Climate' - Setting
[17:16:46][D][climate:040]:   Target Temperature: 21.00
[17:16:46][D][esp-idf:000][mhi_task]: E (2191909) MHI-AC-CTRL-core: Missed 20 frames

[17:16:46][D][sensor:093]: 'Frame errors': Sending state 6955.00000  with 0 decimals of accuracy
[17:16:46][D][esp-idf:000][mhi_task]: W (2191957) MHI-AC-CTRL-core: error 1 . trans len: 165

[17:16:46][D][sensor:093]: 'Frame errors': Sending state 6956.00000  with 0 decimals of accuracy
[17:16:47][D][esp-idf:000][mhi_task]: E (2192959) MHI-AC-CTRL-core: Missed 20 frames

[17:16:47][D][sensor:093]: 'Frame errors': Sending state 6976.00000  with 0 decimals of accuracy
[17:16:48][D][esp-idf:000][mhi_task]: W (2193007) MHI-AC-CTRL-core: error 1 . trans len: 172

[17:16:48][D][sensor:093]: 'Frame errors': Sending state 6977.00000  with 0 decimals of accuracy
[17:16:49][D][esp-idf:000][mhi_task]: E (2194009) MHI-AC-CTRL-core: Missed 20 frames

[17:16:49][D][sensor:093]: 'Frame errors': Sending state 6997.00000  with 0 decimals of accuracy
[17:16:49][D][esp-idf:000][mhi_task]: W (2194057) MHI-AC-CTRL-core: error 1 . trans len: 155

[17:16:49][D][sensor:093]: 'Frame errors': Sending state 6998.00000  with 0 decimals of accuracy
[17:16:50][D][esp-idf:000][mhi_task]: E (2195059) MHI-AC-CTRL-core: Missed 20 frames

[17:16:50][D][sensor:093]: 'Frame errors': Sending state 7018.00000  with 0 decimals of accuracy
[17:16:50][D][esp-idf:000][mhi_task]: W (2195107) MHI-AC-CTRL-core: error 1 . trans len: 165
```

Turning on (or off) the indoor unit via ESPHome, changing the target temperature, fan mode, etc. does not work. Do you have any clue on why that may be?

nohum, 2025-03-03:
Hi, thanks for your quick response.

The ESP is only used for the AC configuration at the moment. I just ordered a logic analyzer and will try to take a look at the frames that are visible behind the level shifter. I will report back then.

hberntsen, 2025-03-03:
This is a sample trace from my unit:
[image]

Notice the spike on the CS after the CLK.

This is the whole trace: [33byte-example.zip](https://github.com/user-attachments/files/19061020/33byte-example.zip). You should be able to unzip that and open it with PulseView.

nohum, 2025-03-25:
Hi, thanks for your sample trace.

I got my hands on a ESP8266 (an original D1 mini) and tried the code from the other repository (https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/issues) with the board from nodo-shop. So far, I can confirm that my HVAC does not support long frames and the board is working fine with the D1 mini, so I think there should be at least no wiring or level shifter issues involved.

Will continue to investigate this.

Deckard99, 2025-09-21:
Hi nohum,

did you get it working? I'm trying the same pcb (from nodo) and the same ESP (C3 Mini). But using the cs_in_pin on GPIO8. That way it should not require modification of the board, right? I have not yet installed the pcb to the AC, because I'd like to have some confidence that the C3 mini is working at all with this pcb.

Deckard99, 2025-09-22:
Well, thats not what I'd liked to hear. ;) I guess I'll give it a try as soon as I find time to open the AC.