hberntsen/mhi-ac-ctrl-esp32 #17: Question about active_mode auto-enabling in ESPHome

xangin, 2025-03-21:
Dear @hberntsen ,

First of all, thank you for developing this integration! It has been working great overall. However, I encountered an issue where the `active_mode` switch automatically turns off when the module is powered on.

I noticed that it can be enabled using a blueprint, but I would prefer to have it automatically enabled within ESPHome itself, so I don’t have to rely on a blueprint.

I have tried using `restore_mode`, but it seems that this parameter is not supported. I also attempted to turn on the switch during boot, but I received the following error:

    "ID 'myswitch1' of type MhiAcCtrl doesn't inherit from switch_::Switch. Please double-check your ID is pointing to the correct value."

Is there any other way to achieve this? I would really appreciate your guidance.

Thank you in advance for your help!

hberntsen, 2025-03-21:
Note that the activation via Home Assistant has advantages:
1. ESPHome has a 15 minute auto reboot when there is no connection. On reboot the active mode is off so the AC will also turn off by itself. That prevents that the AC runs when you can't control it any more.
2. While the network/Home Assistant is down you can use timers with the remote again while you are fixing things :).

Though, if you still want to activate it on the device, this works for me:

```yaml 
esphome:
  on_boot:
    then:
      - switch.turn_on: ${device_id}_active_mode

switch:
  - platform: MhiAcCtrl
    mhi_ac_ctrl_id: ${mhi_device_id}
    active_mode:
      name: ${device_name} Active mode
      id: ${device_id}_active_mode
```

I've also added support for `restore_mode` for it, so this should work as well now:

```yaml
switch:
  - platform: MhiAcCtrl
    mhi_ac_ctrl_id: ${mhi_device_id}
    active_mode:
      name: ${device_name} Active mode
      id: ${device_id}_active_mode
      restore_mode: ALWAYS_ON
```

xangin, 2025-03-22:
Hi @hberntsen ,

Thank you for your detailed explanation and for adding support for `restore_mode`! 

I really appreciate you taking the time to modify the code specifically for my request. Your support means a lot!

I have implemented the changes based on your suggestions, and they work perfectly for my needs. Everything is now functioning exactly as I had hoped.

Once again, thank you so much for your help and for your amazing work on this project! 😊

hberntsen, 2025-03-28:
Thank you for your response! It is really nice to hear my work was valuable to you!

xangin, 2025-04-09:
Hi @hberntsen ,

I have a quick question regarding the `status_led` behavior.

Does your code include handling for `status_led`?
If the module has already been added to Home Assistant but the air conditioner hasn't been physically connected yet, will the `status_led` keep blinking until the AC is properly connected?

Just wanted to confirm how this part behaves. Thank you again! 😊

harm-nedap, 2025-04-12:
Hi @xangin 

For #10 I added some handling: https://github.com/search?q=repo%3Ahberntsen%2Fmhi-ac-ctrl-esp32-c3%20status_&type=code . I'm not using the status led myself (not visible because my module is inside the AC), so this is untested. It should blink when the AC is not connected!

hberntsen, 2025-04-12:
Hi @xangin 

For #10 I added some handling: https://github.com/search?q=repo%3Ahberntsen%2Fmhi-ac-ctrl-esp32-c3%20status_&type=code . I'm not using the status led myself (not visible because my module is inside the AC), so this is untested. It should blink when the AC is not connected!

xangin, 2025-04-12:
Hi @hberntsen ,

Thank you for the clarification and for pointing me to the code reference!

Good to know that the `status_led` is intended to blink when the AC is not connected — that’s exactly the behavior I was curious about. Even if it’s untested, it’s helpful to understand how it should work.

I really appreciate your continued support. Thanks again!

hberntsen, 2026-09-08:
I've put the new method of my post above in the example. Feel free to reopen when you have more to say :)

closed 2026-09-08 by hberntsen