hberntsen/mhi-ac-ctrl-esp32 #21: Compatibility issue with ESPHome 2025.6.3 – custom_component fails with xTaskCreatePinnedToCore assert

xangin, 2025-07-13:
Hi @hberntsen,

Thank you so much for your work on this integration! It’s been working perfectly for me on ESPHome 2024.12.4, but I encountered a compatibility issue with 2025.6.3.

📝 Issue Summary

When using ESPHome 2025.6.3 on an ESP32-C3, the device fails to boot properly and enters safe mode. The status_led also changes from slow blinking to **rapid blinking**.

The log shows the following critical error:

`assert failed: xTaskCreatePinnedToCore freertos_tasks_c_additions.h:162`

This seems related to ESP-IDF 5.x changes where `xTaskCreatePinnedToCore()` requires a valid core ID. Since ESP32-C3 is single-core, using this API may trigger an assertion.

When rolling back to ESPHome 2024.12.4 (ESP-IDF 4.x), the same configuration works flawlessly.
📄 Attached

I’ve attached the full fail log for your reference. [fail_log_20250603.txt](https://github.com/user-attachments/files/21204163/fail_log_20250603.txt)

✅ Steps to Reproduce

1. Flash the ESP32-C3 with ESPHome 2025.6.3 and the current MhiAcCtrl configuration.

2. Boot the device.

3. Observe rapid blinking status_led and safe mode entry.

🙏 Request

Could you check if the component needs adjustments for ESP-IDF 5.x and ESPHome 2025.x compatibility (especially regarding xTaskCreatePinnedToCore)? I’d be happy to help test any updated version.

Thanks again for your time and for this excellent project!

Best regards,

xangin, 2025-07-16:
Hi @hberntsen ,

Thank you for the quick fix! I’ve tested the updated code from the master branch on ESPHome 2025.6.3, and everything is working perfectly now. The device no longer enters safe mode, and all functions (including `status_led`) behave as expected.

By the way, I also designed my own hardware based on your project, and it works great with your software. I’ve shared my hardware design on GitHub here: [Hardware.md](https://github.com/xangin/mhi-ac-ctrl-esp32-c3/blob/master/Hardware.md)

The documentation is in Chinese, but you can still reference the pictures if you’re interested.

Once again, thank you so much for this amazing project! 🚀

closed 2025-07-16 by xangin

hberntsen, 2025-07-16:
Glad to hear it works! Also nice work with your hardware design! I like the enclosure you made for it :)

xangin, 2025-07-17:
Thanks! It's my honor to be part of your repository. 💯

HpNoTiQ56, 2025-10-01:
Hi Xangin,
I'm bought a ESP32-C6 supermini and i'm looking to use your config :
Could you share your 3d print files and add how to add an external antenna. You say to remove old smd antenna but it's still there in photo : is this a recommandation or mandatory?

xangin, 2025-10-02:
> Hi Xangin, I'm bought a ESP32-C6 supermini and i'm looking to use your config : Could you share your 3d print files and add how to add an external antenna. You say to remove old smd antenna but it's still there in photo : is this a recommandation or mandatory?

Sure! I have uploaded the STL file here: [https://github.com/xangin/mhi-ac-ctrl-esp32-c3/blob/master/mhi-case-hole.stl](https://github.com/xangin/mhi-ac-ctrl-esp32-c3/blob/master/mhi-case-hole.stl)

Whether to add an external antenna depends on the module's layout. I've noticed that the ESP32-C6 you're using doesn't have an IPEX connector, so an external antenna is not necessary.