hberntsen/mhi-ac-ctrl-esp32 #22: Wrong MOSI signature using old PCB & ESP32 Live mini Kit

HpNoTiQ56, 2025-07-20:
Hi,
I'm struggling to get HW SPI working with my conf.
My A/C are MHI. I'm using absalum PCB 2.1 with level shifters. They are working great with ESP8266 since 2021.
I migrated one of my unit to use ESP32 (Live mini kit ~ Wemos mini esp32). It's working great with SW SPI (ginkage) by adapting PINs.
https://doc.riot-os.org/group__boards__esp32__mh-et-live-minikit.html#esp32_mh_et_live_minikit_pinout
MISO : 19
MOSI : 23
SCK : 18

Until now, I could use Bluetooth proxy with some loop errors but now I can't compile since 4.1 because of size of Arduino. That's why I'm trying to move in HW SPI.

I've seen that I need to connect 2 GPIO to fake CS_IN & CS_OUT : I did on my PCB bridging D3 & D4 - GPIO 16 & 17 regarding Datasheet.
https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Hardware.md

I can control AC but status won't change in HA. Logs are flouded with : 

[22:10:33][I][esp-idf:000][mhi_task]: W (36624) MHI-AC-CTRL-core: wrong MOSI signature. 0x49 0x00 0x00
[22:10:33][I][esp-idf:000][mhi_task]: W (36624) MHI-AC-CTRL-core: error -1 . trans len: 264

Could you help me work it out?

HpNoTiQ56, 2025-07-21:
Hi, thank you for your quick answer.

- 1048 -> Not booting.
- 3096 -> [22:19:20][I][esp-idf:000][mhi_task]: W (14614) MHI-AC-CTRL-core: error 1 . trans len: 0
[22:19:20][I][esp-idf:000][mhi_task]: I (14614) MHI-AC-CTRL-Operation-Data: Timeout when requesting Operation Data SetTemperature

hberntsen, 2025-07-21:
Some messages during startup are ok. I've seen some on my units as well with switching from 20 to 33 bytes. Do you get status updates now?

HpNoTiQ56, 2025-07-21:
I've got flooded error 1 using use_long_frame: false.
Now, using 4096 

[image]

Using 16484 

[image]

hberntsen, 2025-07-22:
@unregist are you still using this project on your ESP32? Does the latest version still work?

@HpNoTiQ56 Could you try branch `issue-22`, it will print the whole frame, maybe that contains some clues.

On top of that: Try to set the DMA channel to `SPI_DMA_CH1` or `SPI_DMA_CH2`: https://github.com/hberntsen/mhi-ac-ctrl-esp32-c3/blob/9197cc12d0f118bd49751cf96e2cfbebb1a93f6a/esphome/components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp#L588C60-L588C75 . The documentation of the ESP-IDF says that specifically for the ESP32: https://github.com/espressif/esp-idf/blob/f5c3654a1c2d2a01f7f67def7a0dc48e691f63c0/docs/en/api-reference/peripherals/spi_slave.rst?plain=1#L77-L79

HpNoTiQ56, 2025-07-22:
Thank you again for trying to help me. 
Here are the full logs : 
`0x23 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0x00 
[21:13:52][I][esp-idf:000][mhi_task]: W (17642) MHI-AC-CTRL-core: error -1 . trans len: 264
[21:13:52][I][esp-idf:000][mhi_task]: W (17689) MHI-AC-CTRL-core: wrong MOSI signature. 0x49 0x00 0x00
[21:13:52][I][esp-idf:000][mhi_task]: W (17691) MHI-AC-CTRL-core: Frame: 0x49 0x00 0x00 0xC0 0x00 0x20 0x1F 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0xFF 0x00 0x00 0x00 0x23 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0x00 
[21:13:52][I][esp-idf:000][mhi_task]: W (17692) MHI-AC-CTRL-core: error -1 . trans len: 264
[21:13:52][I][esp-idf:000][mhi_task]: W (17739) MHI-AC-CTRL-core: wrong MOSI signature. 0x49 0x00 0x00
[21:13:52][I][esp-idf:000][mhi_task]: W (17741) MHI-AC-CTRL-core: Frame: 0x49 0x00 0x00 0xC0 0x00 0x20 0x1F 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0xFF 0x00 0x00 0x00 0x23 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0x00 
[21:13:52][I][esp-idf:000][mhi_task]: W (17741) MHI-AC-CTRL-core: error -1 . trans len: 264
[21:13:52][I][esp-idf:000][mhi_task]: W (17789) MHI-AC-CTRL-core: wrong MOSI signature. 0x49 0x00 0x00
[21:13:52][I][esp-idf:000][mhi_task]: W (17792) MHI-AC-CTRL-core: Frame: 0x49 0x00 0x00 0xC0 0x00 0x20 0x1F 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0xFF 0x00 0x00 0x00 0x23 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0x00 
[21:13:52][I][esp-idf:000][mhi_task]: W (17792) MHI-AC-CTRL-core: error -1 . trans len: 264
[21:13:52][I][esp-idf:000][mhi_task]: W (17839) MHI-AC-CTRL-core: wrong MOSI signature. 0x49 0x00 0x00
[21:13:52][I][esp-idf:000][mhi_task]: W (17841) MHI-AC-CTRL-core: Frame: 0x49 0x00 0x00 0xC0 0x00 0x20 0x1F 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0xFF 0x00 0x00 0x00 0x23 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0x00 
[21:13:52][I][esp-idf:000][mhi_task]: W (17842) MHI-AC-CTRL-core: error -1 . trans len: 264
[21:13:52][I][esp-idf:000][mhi_task]: W (17889) MHI-AC-CTRL-core: wrong MOSI signature. 0x49 0x00 0x00
[21:13:52][I][esp-idf:000][mhi_task]: W (17891) MHI-AC-CTRL-core: Frame: 0x49 0x00 0x00 0xC0 0x00 0x20 0x1F 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0xFF 0x00 0x00 0x00 0x23 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0x00 
[21:13:52][I][esp-idf:000][mhi_task]: W (17892) MHI-AC-CTRL-core: error -1 . trans len: 264
[21:13:52][I][esp-idf:000][mhi_task]: W (17939) MHI-AC-CTRL-core: wrong MOSI signature. 0x49 0x00 0x00
[21:13:52][I][esp-idf:000][mhi_task]: W (17941) MHI-AC-CTRL-core: Frame: 0x49 0x00 0x00 0xC0 0x00 0x20 0x1F 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0xFF 0x00 0x00 0x00 0x23 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0x00 
[21:13:52][I][esp-idf:000][mhi_task]: W (17942) MHI-AC-CTRL-core: error -1 . trans len: 264
[21:13:52][I][esp-idf:000][mhi_task]: W (17989) MHI-AC-CTRL-core: wrong MOSI signature. 0x49 0x00 0x00
[21:13:52][I][esp-idf:000][mhi_task]: W (17991) MHI-AC-CTRL-core: Frame: 0x49 0x00 0x00 0xC0 0x00 0x20 0x1F 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0xFF 0x00 0x00 0x00 0x23 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0x00 
[21:13:52][I][esp-idf:000][mhi_task]: W (17991) MHI-AC-CTRL-core: error -1 . trans len: 264
[21:13:52][I][esp-idf:000][mhi_task]: W (18039) MHI-AC-CTRL-core: wrong MOSI signature. 0x49 0x00 0x00
[21:13:52][I][esp-idf:000][mhi_task]: W (18041) MHI-AC-CTRL-core: Frame: 0x49 0x00 0x00 0xC0 0x00 0x20 0x1F 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0xFF 0x00 0x00 0x00 0x23 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0x00 
[21:13:52][I][esp-idf:000][mhi_task]: W (18042) MHI-AC-CTRL-core: error -1 . trans len: 264
[21:13:52][I][esp-idf:000][mhi_task]: W (18089) MHI-AC-CTRL-core: wrong MOSI signature. 0x49 0x00 0x00
[21:13:52][I][esp-idf:000][mhi_task]: W (18091) MHI-AC-CTRL-core: Frame: 0x49 0x00 0x00 0xC0 0x00 0x20 0x1F 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0xFF 0x00 0x00 0x00 0x23 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0x00 
[21:13:52][I][esp-idf:000][mhi_task]: W (18092) MHI-AC-CTRL-core: error -1 . trans len: 264
[21:13:52][I][esp-idf:000][mhi_task]: W (18139) MHI-AC-CTRL-core: wrong MOSI signature. 0x49 0x00 0x00
[21:13:52][I][esp-idf:000][mhi_task]: W (18141) MHI-AC-CTRL-core: Frame: 0x49 0x00 0x00 0xC0 0x00 0x20 0x1F 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0xFF 0x00 0x00 0x00 0x23 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0x00 
[21:13:52][I][esp-idf:000][mhi_task]: W (18141) MHI-AC-CTRL-core: error -1 . trans len: 264
[21:13:52][I][esp-idf:000][mhi_task]: W (18189) MHI-AC-CTRL-core: wrong MOSI signature. 0x49 0x00 0x00
[21:13:52][I][esp-idf:000][mhi_task]: W (18191) MHI-AC-CTRL-core: Frame: 0x49 0x00 0x00 0xC0 0x00 0x20 0x1F 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0xFF 0x00 0x00 0x00 0x23 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0x00 
[21:13:52][I][esp-idf:000][mhi_task]: W (18192) MHI-AC-CTRL-core: error -1 . trans len: 264
[21:13:52][I][esp-idf:000][mhi_task]: W (18239) MHI-AC-CTRL-core: wrong MOSI signature. 0x49 0x00 0x00
[21:13:52][I][esp-idf:000][mhi_task]: W (18241) MHI-AC-CTRL-core: Frame: 0x49 0x00 0x00 0xC0 0x00 0x20 0x1F 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0xFF 0x00 0x00 0x00 0x23 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0x00 
[21:13:52][I][esp-idf:000][mhi_task]: W (18242) MHI-AC-CTRL-core: error -1 . trans len: 264
[21:13:52][I][esp-idf:000][mhi_task]: W (18289) MHI-AC-CTRL-core: wrong MOSI signature. 0x49 0x00 0x00
[21:13:52][I][esp-idf:000][mhi_task]: W (18291) MHI-AC-CTRL-core: Frame: 0x49 0x00 0x00 0xC0 0x00 0x20 0x1F 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0xFF 0x00 0x00 0x00 0x23 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0x00 `

Trying to change channel and edit here.

unregist, 2025-07-24:
> [@unregist](https://github.com/unregist) are you still using this project on your ESP32? Does the latest version still work?

No longer using ESP32. I had a mix of ESP32 and ESP32-C3 and grew tired of managing different configurations. I'm fully ESP32-C3 now.

HpNoTiQ56, 2025-07-24:
Hi, 
I didn't get the trace on my unit.
Here is a log using very_verbose.
Unit is responding well (vanes/mode) but can't power off unit for instance since it's locked to power off. 
My problem is that no value are send from unit to ha. I've got 0 sensor working(temp/vane position/...)

[logs_esp-clim-salon-192-168-1-201_logs(4).txt](https://github.com/user-attachments/files/21409828/logs_esp-clim-salon-192-168-1-201_logs.4.txt)